chipsalliance/Caliptra #13: Acronyms

steven-bellock, 2023-02-28:
### 1
The following acronyms are not aligned with the specifications they are presumably derived from.
- `Composite Device Identifier` -> `Compound Device Identifier` (TCG DICE)
- `Certificate Authority` -> `Certification Authority` (rfc5280)

### 2
Can `FW First Mutable Code` just be `First Mutable Code`?

### 3
The following acronyms are not referenced in the body of the specification. Can we delete them from the acronyms table?
- MCTP
- TCI
- UECC

### 4
CSP is defined as `Critical Security Parameter` however its use in the text is not consistent with that.
> A well and narrowly defined specification maximizes architectural composability, reusability across CSPs, products and vendors, and feasibility of open sourcing.

steven-bellock, 2023-02-28:
### 5
What is `RMC`?

bluegate010, 2023-02-28:
Thanks Steven - I believe `RMC` should read `FMC`. And it looks like `CSP` is used as if it were defined as `Cloud Service Provider`. Yes to everything else.

PR #15 (2023-03-01): Fix acronyms
Fixes #13.
commit 8fe6e158 (2023-03-01): Fix acronyms (#15)

Fixes #13.
commit 2392cdcc (2023-08-27): Fix acronyms (#15)

Fixes #13.
closed 2023-03-01 by andreslagarcavilla