chipsalliance/Caliptra #19: Inconsistent/incorrect encoding for security state

mahmud2023, 2023-03-03:
The security state coded in Table 7 (v 0.62) of [Caliptra.md](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md) uses the same encoding for Provisioned and Manufacturing life cycle states:

- 000b: DebugUnlocked & Unprovisioned
- 100b: DebugLocked & Manufacturing
- 101b: DebugLocked & Production
- 001b: DebugUnlocked & Production

The leftmost bit represents DEVICE_LOCKED status (0 => unlocked, 1 => locked), so lower 2-bits cannot both be 00 for UNPROVISIONED as well as MANUFACTURING.

Bryankel, 2023-03-09:
"000b: DebugUnlocked & Unprovisioned" encoding should be changed, it's a target for undervoltage attacks.

andreslagarcavilla, 2023-09-01:
Bharat, can you resolve this in the spec by 9/8? Than hand off to Stephanie. Thanks

varuns-nvidia, 2023-10-05:
I don't see this as an issue. the leftmost bit is Debug state, not device unlocked state.
I filed an issue for Bryan's comment.

closed 2023-10-05 by varuns-nvidia