chipsalliance/Caliptra #20: Define command for retrieving Caliptra firmware version

bluegate010, 2023-03-06:
May choose to represent FMC+RT fw versions as hashes, rather than numerical values. This command should also include SVN information.

jhand2, 2023-05-19:
We have registers to 2 4-byte words. This could presumably be used for reporting an FMC version and an RT version: https://chipsalliance.github.io/caliptra-rtl/main/internal-regs/?p=clp.soc_ifc_reg.CPTRA_FW_REV_ID%5B0%5D

But if we also need to report SVNs, maybe a mailbox command is still necessary.

andreslagarcavilla, 2023-09-01:
Jeff, please close on this. Thanks

jhand2, 2023-09-01:
This is implemented in code. The image header reports versions and SVNs, which we expose over registers & the FW_INFO mailbox command. But we do still need to document this behavior.

FerralCoder, 2023-10-06:
moving to sw repo where this doc exists