chipsalliance/Caliptra #21: UDS Derivation

steven-bellock, 2023-03-06:
The specification says, regarding UDS derivation:
>The Caliptra UDS is stored in fuses, and is encrypted at rest by an obfuscation secret[2](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#user-content-fn-2-068314cb52189d92e7c1de88e6e6a6df) known only to the Caliptra ROM. Once read by Caliptra ROM at boot, the UDS is then used to derive the IDevID identity.

and
>This obfuscation secret may be a chip-class secret, or a chip-unique PUF, with the latter preferred.

Two questions / issues.
1. It seems like the value that is stored in fuses is the UDS Seed, which, after de-obfuscation becomes the UDS. In that case "The Caliptra UDS is stored in fuses," should be "The Caliptra UDS Seed is stored in fuses,"
2. If I have a PUF why can't I skip the fuses and just use the PUF value, or a secure derivate of it, as the UDS?

bluegate010, 2023-03-13:
I think s/UDS/UDS seed/ makes sense.

As for PUF, it probably depends on what kind of PUF. An SRAM PUF requires a large error correction vector to be written to persistent storage. I believe such data tends to be significantly larger than a straight-up UDS, _but_ it's non-secret, so doesn't need as careful handling as UDS.

One other thing to consider is FIPS zeroization requirements. Assume a PUF technology that requires no error correction vector. How would we satisfy zeroization? Would we need to lobby NIST to relax those guidelines in certain cases?

JohnTraverAmd, 2023-03-28:
To be able to zero this for FIPS, you would want to have both a PUF and a fuse value that together creates the UDS.  This way you are able to still zero it.

Therefore, our current direction is still correct.  The PUF can supply the "obfuscation secret" while the fuses contain the UDS seed.

steven-bellock, 2023-03-30:
@JohnTraverAmd Could one use (at least) a single fuse, or some other destructive means, to permanently deny power to the PUF as a means to zeroize it?

bluegate010, 2023-10-27:
> @JohnTraverAmd Could one use (at least) a single fuse, or some other destructive means, to permanently deny power to the PUF as a means to zeroize it?

Fwiw this was raised, very briefly, in a talk at ICMC this year as part of an argument for why PUFs are certifiable. There didn't seem to be consensus on this point.

varuns-nvidia, 2023-10-31:
@steven-bellock is this good to close?

steven-bellock, 2023-10-31:
I am still bummed about PUFs, but yes this can be closed.

closed 2023-10-31 by steven-bellock