chipsalliance/Caliptra #27: UDS transmission across a "secure channel"

bluegate010, 2023-03-13:
Raised by Jeff Plank. To achieve FIPS certification, we will need to demonstrate that UDS (or UDS seed, depending on how #21 shakes out) is transmitted into Caliptra across a "secure channel". [SP 800-90A](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-90Ar1.pdf) defines secure channels as follows:

> A path for transferring data between two entities or components that ensures confidentiality, integrity and replay protection, as well as mutual authentication between the entities or components. The secure channel may be provided using **approved** cryptographic, physical or procedural methods, or a combination thereof. Sometimes called a trusted channel.

So the fuse registers that ferry UDS from the SoC's fuse bank into Caliptra will need to adhere to that definition. What shall be the narrative here? Can we claim that confidentiality, integrity, and replay protection are provided by "physical methods"?

andreslagarcavilla, 2023-09-01:
John, is this still an open?

varuns-nvidia, 2023-10-05:
The fuse block must be part of the cryptographic module boundary, so it is not transmitted across a secure channel. @JohnTraverAmd please correct if otherwise.

closed 2023-10-05 by varuns-nvidia

JohnTraverAmd, 2023-10-06:
Varun was correct.  It isn't a trusted channel, the fuse box would be considered part of the FIPS module.