chipsalliance/Caliptra #42: Complete and merge DPE + Quote API into spec.md

andreslagarcavilla, 2023-04-07:
Let's complete this so 0p8 hw changes can be called out

andreslagarcavilla, 2023-09-01:
@jhand2 @vsonims @bluegate010   is this now part of spec.md? I'll assign to you for closure by 9/8. Thanks

jhand2, 2023-09-01:
This is blocked on https://github.com/chipsalliance/caliptra-sw/pull/429. @vsonims please review.

varuns-nvidia, 2023-09-20:
@andreslagarcavilla , is it sufficient to
* link to runtime specification in spec.md (done here https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#service-surface )
* have all the APIs documented in caliptra-sw (done here https://github.com/chipsalliance/caliptra-sw/blob/main/runtime/README.md )

varuns-nvidia, 2023-10-05:
I recommend closing this.

jhand2, 2023-10-06:
Agreed, this is complete.

closed 2023-10-06 by jhand2