chipsalliance/Caliptra #48: Finalize IDev_ID_Cert generation and template

jameszhang-nvidia, 2023-05-13:
As we have discussed during the Dec F2F last year, we have decided to create the immutable identity that is tied to silicon (IDev_ID_Cert) with templating and burning the certificate's signature into FUSE.  Currently the General Spec and ROM Spec calls out this as "FUSE_IDEVID_CERT_ATTR" which list it as information that is used to generate IDev_ID_CSR.  This is not how it should be used.  As we discussed in Today's (5/12/2023) Caliptra FW weekly that we agree we need a determnistically generated IDev_ID_Cert during runtime.  IDev_ID_CSR is only done during Silicon Vendor Manufacture time so we do not need to store said information in fuse.

IDev_ID_Cert is also something we would not need to generate during the ROM phase.  As all the information should be available either through Templating or the Signature in FUSE, it can be generated later in mutable code.  One thing that people raised was since we do not want Caliptra integration to change ROM / FMC / FW code at all, so it means templating for IDev_ID_Cert and IDev_ID_CSR can not differ from vendor to vendor.  This is problematic because items such as Serial Number, Subject and Issuer needs to differ from Vendor to Vendor.  Our conversation during the meeting had a strawman where we said that it is possible to include those information in the FW Manifest.  However that won't out for two reasons.

1. The Serial Number and Subject of the cert would be per device unique
2. The IDev_ID_CSR also needs to be generated, and that is done in ROM.  That will likely come without manifest information

So to resolve this I would suggest that there are 2 choices

1. Provide even more fuse so that it will be a SOC manufacture's responsibility to fuse in Serial Number/Issuer and Subject into the fuse and that gets pushed to Caliptra through the fuse push.   Looking at a typical Device_ID_Cert that we have, This would require at probably 150 bytes (1200 bits).  This is a pretty heavy burden so I would suggest option 2
2. Have SOC ROM push those information into Caliptra deterministically through some other means.  This can be register writes or another mailbox command, this would establish vendor specific string for cert template to Caliptra without Caliptra having to implement more FUSE or HW changes.  This is purely a ROM change.

At this point I would suggest Item 2.

andreslagarcavilla, 2023-09-01:
James, is this still open? If so, don't we need to support this via a ROM change?

jameszhang-nvidia, 2023-09-01:
This is resolved with the push of using TBS and allow SOC manager processor to pass in those needed info.

With the closure of https://github.com/chipsalliance/caliptra-sw/issues/294 and https://github.com/chipsalliance/caliptra-sw/pull/709 I believe this overall issue can be closed.

Jordan setup a meeting next week to discuss and close this overall .

jhand2, 2023-09-01:
Once https://github.com/chipsalliance/caliptra-sw/issues/713 is closed, we can close this.

For now, https://github.com/chipsalliance/caliptra-sw/pull/709 takes the signature as a mailbox input. Even if we make no changes in ROM, this mechanism will work. It just requires the SoC manager to handle storing the signature.

commit 5729ec81 (2023-10-04): Document certificates and all evidence reported

Fixes #48
Fixes #57
commit 09d1e0cf (2023-10-05): Document certificates and all evidence reported

Fixes #48
Fixes #57
commit 516ec8f5 (2023-10-06): Document certificates and all evidence reported (#132)

* Document certificates and all evidence reported

Fixes #48
Fixes #57

* swap least and most

* fix the serial number algorithm again
closed 2023-10-06 by andreslagarcavilla