chipsalliance/Caliptra #52: Fuse Loading and Zeroizing Definition

JohnTraverAmd, 2023-05-26:
Description:
Caliptra needs to more clearly describe Fuse Loading and Zeroizing mechanisms that meet Caliptra requirements and are FIPS certifiable.

JohnTraverAmd, 2023-05-26:
Some of the items that need to be clarified.

- Who can access the Caliptra Fuses.
- How Caliptra asks for fuses to be zeroized.
- How Caliptra knows that the fuses have been successfully zeroized.
- How these fuses are programmed in manufacturing.
- How to ensure that the public key hash and security state fuses are correctly translated from fuse to Caliptra IP registers.

andreslagarcavilla, 2023-09-01:
Varun, John can you expand the .md with this information by 9/8? Thanks. Then please handoff to steph-morton for editing.

closed 2023-09-06 by varuns-nvidia