chipsalliance/Caliptra #60: Inconsisent mailbox protocol

ChiaweiW, 2023-06-26:
Dear All,

I found an inconsistne illustration on the lock-releasing flow of mailbox protocol and got below questions.

Q1.
In [Caliptra.md](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md), the lock is released by Receiver.
Whereas in [Caliptra_Integration_Specification](https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/Caliptra_Integration_Specification.pdf), the lock is released by Sender.

Q2.
The specfication states that mailbox should accept writes only from the device that locked the mailbox.
The receiver protocol does not have the lock-acquiring step
However, if there is data to respond, Receiver has to writes DLEN, DATAIN, and STATUS registers in the middle of a transaction, where the lock is still held by Sender.

Could you help me to clarify on these topics? Any guidance is appreciated.

Regards,
Chiawei

korran, 2023-06-26:
The integration specification is correct; the receiver should never touch the lock.

Our integration tests use this register logic (while acting as the SoC) for sending mailbox commands to Caliptra and receiving the response:

https://github.com/chipsalliance/caliptra-sw/blob/965f264299bf74c256da023cf95518f7ca96efec/hw-model/src/lib.rs#L480-L538

Leaving this bug open until Caliptra.md is updated with the correct logic.

ChiaweiW, 2023-06-27:
Hi @korran

Thanks for clarifying Q1. Will keep this open until Caliptra.md is revised.

Regarding the Q2, after tracing the source, it appears that the lock does not physically (i.e. by HW) block Receiver from writing DLEN, STATUS, and DATAIN registers. Will this result in (un)intentional race conditions between Caliptra and SoC?

calebofearth, 2023-06-27:
@ChiaweiW 
There are a couple of protections in the hardware against invalid writes to DLEN, STATUS, DATAIN registers.
(1) Internal logic enforces write privilege using the signals `valid_receiver` and `valid_requester` in [mbox.sv](https://github.com/chipsalliance/caliptra-rtl/blob/main/src/soc_ifc/rtl/mbox.sv) - if the privilege conditions are not met for writes to these registers, the write is dropped and registers are not impacted.
(2) There is a strict ordering requirement for mailbox register accesses to execute the mailbox flow. If this procedure is violated, the mbox_protocol error signal triggers a cptra_error_non_fatal interrupt ( to the soc), soc_ifc_error_intr (internal intr to the RV uC), and forces a state transition to the ERROR state in the mailbox, which is only recovered by a force unlock from the uC.

ChiaweiW, 2023-06-28:
@calebofearth 

Thanks for the explanation and show me the HW implementation!

andreslagarcavilla, 2023-09-01:
Kor, can you update the .md by 9/8 and then hand off to steph-morton for refinements? Thanks

calebofearth, 2023-09-07:
Didn't see that you assigned this to Kor - I have the fix ready to go in #79

closed 2023-10-05 by varuns-nvidia