chipsalliance/Caliptra #61: Makefile Error : trying to build program.hex file

ax-ag16, 2023-06-28:
I am trying to make Makefile on linux-cent-OS  (7.9.2009).

 I am trying to follow the README.md but i am unable to get past this error. I am able to see the defines.h file in the path but its not getting copied to run directory

`make -f ${CALIPTRA_ROOT}/tools/scripts/Makefile 
rm -rf *.log *.s *.hex *.dis *.size *.tbl irun* vcs* simv* .map *.map snapshots \
verilator* *.exe obj* *.o ucli.key vc_hdrs.h csrc *.csv work \
dataset.asdb  library.cfg vsimsa.cfg  riviera-build wave.asdb sim.vcd \
*.h
make: *** No rule to make target `/mypath/caliptra-rtl-0.8/run/defines.h', needed by `program.hex'.  Stop.`

calebofearth, 2023-06-29:
Issues of this nature should be filed over at [caliptra-rtl](https://github.com/chipsalliance/caliptra-rtl/issues) instead of this repo -- for future reference.

Please post the following to help debug:
1. current working directory when running the script
2. Exact command used to trigger this error
3. Value of env variables described in [README.md](https://github.com/chipsalliance/caliptra-rtl/blob/main/README.md#environment-variables)

This error usually happens because of some environment issue, like undefined or bad values for variables.

ax-ag16, 2023-06-30:
Able to resolve. Thank you

ax-ag16, 2023-06-30:
Able to resolve .thank you!

closed 2023-06-30 by ax-ag16