chipsalliance/Caliptra #62: Consolidate Caliptra Mailbox Commands in the Main Spec

fdamato, 2023-06-30:
Caliptra Mailbox Commands are the main interface for an integrator to look at.
As of now the most updated description of the Caliptra MB commands resides in the individual components spec (ROM, RT FW).

We need to 
   1. Pull out the Description of the Caliptra MB commands from the individual component specs
   2. Populate the main Caliptra Spec with the correct and most updated description of the Mailbox Commands
   3. Link each individual component mailbox command description to the main Spec

andreslagarcavilla, 2023-09-01:
Jordan and Vishal -- please complete this by 9/8. Then hand off to steph-morton. Thanks

jhand2, 2023-09-01:
https://github.com/chipsalliance/Caliptra/pull/71 updates this spec to point to the RT spec, which has up-to-date command definitions. If we thing that is sufficient, we can close this issue once that PR is merged.

varuns-nvidia, 2023-10-05:
Closing this, @fdamato feel free to re-open if you feel otherwise

closed 2023-10-05 by varuns-nvidia