chipsalliance/Caliptra #63: HW Release Model - Collaterals Test

fdamato, 2023-07-07:
The HW Model Release Documentation should define a list of recommended tests that the integrator should execute to validate the Caliptra RTL in the SOC environment.
 
Those tests should be delivered as part of the release collateral archive

andreslagarcavilla, 2023-09-01:
Fabrizio, please assign to steph-morton when done. Can you provide this by 9/8? Thanks

andreslagarcavilla, 2023-09-01:
Sorry, I realize there is a PR merged. Close the issue? Or more work needed?

FerralCoder, 2023-09-06:
@andreslagarcavilla, the merged PR that was referenced above does not address this issue. A few of us at AMD have started creating this test document this week, but it will not be done by 9/8. We intend to bring the framework to present to the larger WG this Friday, and hopefully get some owners assigned to help build out the details collaberatively.