chipsalliance/Caliptra #65: Documentation refers to the RTL, ROM, and FW as reference designs, which is misleading and incorrect.

andrewsil1, 2023-07-20:
As per the title, we have not made it sufficiently clear that the Caliptra implementations that we are providing are not traditional "references" which are free to be forked and modified by the user community, but rather that while they are open source they are meant to be used and adopted as-is, in an unmodified fashion, in order to maintain the guarantees of industry-wide consistency of the logic behavior, and predictable signatures of the blessed ROM and FMC contents.

We should update the documentation to make this expectation more clear, as I am still receiving questions from prospective adopters that show that there is still confusion on this point.

andreslagarcavilla, 2023-09-01:
Stephanie please assign to me once you've restructured the spec and have a placeholder section to clarify this.

steph-morton, 2023-09-25:
Spec is restructured. Recommend you place this clarification after the third para of the [Silicon RoT goals](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#silicon-rot-goals) section.

varuns-nvidia, 2023-10-31:
Closing this with #147

closed 2023-10-31 by varuns-nvidia