chipsalliance/Caliptra #67: Area/Complexity of Caliptra in Integration guide

fheaton59, 2023-08-15:
There is a table in the integration guide that provides area information for a Caliptra wrapper obtained from Design Compiler. 
Are the area numbers in units of square microns?  
If so, what technology was used to perform the run that generated the results?  
In addition, what does the wrapper cell include and not include?
	I assume none of the imem, dmem, or ROM blocks are included. Is that right?
	Does it include the RISC V processor?

bharatpillilli, 2023-10-06:
But RISC-V is included in the numbers. 

Can't tag @nstewart-amd as he added these numbers from AMD's syn runs

 @pkwidzin-amd (can u add him plz?)

nstewart-amd, 2023-10-06:
These metrics are inclusive of VeeR core, Caliptra logic, imem/dmem RAM, ROM.
The area is expressed in units of square microns.
The target foundry technology node is an industry standard, moderately advanced technology node as of 2023 September.

fheaton59, 2023-10-13:
> <<< External Email >>>
>
> Can this be closed?
>
> -
>
> ** This message and any attachments are for the sole use of the intended recipient(s). It may contain information that is confidential and privileged. If you are not the intended recipient of this message, you are prohibited from printing, copying, forwarding or saving it. Please delete the message and attachments and notify the sender immediately. **
>
> Rambus Inc.<http://www.rambus.com>

Yes, I appreciate the answer to my questions. Maybe I do not have permissions but I could not find a way through my web interface to the github issue to directly add a comment or to close this issue. Could someone who has better accessibility than I please close the issue please.

closed 2023-10-13 by FerralCoder