chipsalliance/Caliptra #68: Update documentation with new ROM size

jhand2, 2023-08-15:
The Caliptra ROM size has increased from 32K -> 48K. But the spec (including diagrams) still has the old values. We should update this.

steph-morton, 2023-09-25:
@andreslagarcavilla please confirm scope: update `ROM (32KB)` to `ROM (48KB)` in following figures: 

- [Figure 2](https://github.com/chipsalliance/Caliptra/blob/main/doc/images/Caliptra_HW_diagram.png)
- [Figure 5](https://github.com/chipsalliance/Caliptra/blob/main/doc/images/Caliptra_HW_Block_diagram.png)
- [Figure 9](https://github.com/chipsalliance/Caliptra/blob/main/doc/images/Caliptra_HW_diagram.png)

If there are associated text updates to make, please list them.

andreslagarcavilla, 2023-09-25:
Yes, those three are necessary changes to make

steven-bellock, 2023-09-25:
Also please use `48 KiB`.

closed 2023-09-27 by andreslagarcavilla