chipsalliance/Caliptra #70: Provide scope for requirements

steven-bellock, 2023-08-18:
There are some requirements which only apply at certain stages of the device's lifecycle. For example 
>If JTAG is enabled pre or post SoC reset, a Calpitra RTM’s dedicated Fuses shall not be accessible (shall not be readable, shall not be writable) by a Caliptra RTM or other SoC IP. This restriction shall be applicable to a Caliptra RTM’s Fuse shadow registers as well (refer to Physical Attack Countermeasures).

may apply only when the device is in Production. It would be good to state at the beginning of the specification that
>Unless otherwise stated requirements are applicable to all stages of the lifecycle.

and then limit the scope of a requirement if it is only applicable in certain stages.

andreslagarcavilla, 2023-09-01:
Bharat, please complete PR with these edits by 9/8. Thanks

bharatpillilli, 2023-09-01:
Hi Andres -  I did not write that requirement. Not sure where it came from. Looping @varuns-nvidia and @pkwidzin-amd to see if they know.

closed 2023-09-06 by varuns-nvidia