chipsalliance/Caliptra #95: Clarify RTM/RTD/RTI and relation to SP 800-193

bluegate010, 2023-09-15:


FerralCoder, 2023-10-06:
@bluegate010, what did you have in mind here? Are you going to make this update? Or if we sync up on what you were thinking, I might be able to take this.

bluegate010, 2023-10-27:
On review of [SP 800-193](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-193.pdf), what it's saying about RTD isn't _really_ about remote attestation, even though we may occasionally use it colloquially that way. For example:

* "Either as an alternative to integrity checks (for devices that cannot support such a capability) or in addition to those checks, the RTD or CTD **may** use watchdog timers to detect potential corruption of critical data."
* "If corruption of critical data is detected, the RTD or CTD **shall** be capable of starting a recovery process to restore the device’s critical data."
* "The detection mechanism **should** be capable of creating notifications of corruption"

So RTD seems more about introspection and allowing a platform to fix itself, rather than about using attestation to provide an outward expression of the platform's current state.

We could maybe say Caliptra is _related_ to RTD, in that it helps an external entity detect firmware corruption. But in my view at least, SP 800-193's RTD is solving a different problem than Caliptra was designed for.

varuns-nvidia, 2023-10-27:
Agree, for Caliptra to meet SP 800-193's intention of an RTD, it needs to actively control the boot process of a device.

The other angle to address in the spec is RTM, which TCG defines as "An RoT that makes the initial integrity measurement, and adds it to a tamper-resistant log." "Initial" is contextual; Caliptra is RTM for itself, but cannot make claims that it made the initial measurement for the SoC as a whole.

FerralCoder, 2023-10-31:
likewise, we could say that Caliptra is RTD and RTU for itself.

bluegate010, 2023-10-31:
RTD / RTM for itself: yes.
RTU for itself: eh. It does enforce anti-rollback, but it doesn't control how Caliptra firmware updates get applied, it just attempts to boot whatever it's given.

FerralCoder, 2023-10-31:
Yes agreed, I had hitless update in mind when I said RTU, but since that is just an ephemeral image that has not necessarily been stored and protected (and if it is was done by someone else) then it does not meet the spirit of RTU

FerralCoder, 2023-10-31:
I put up PR https://github.com/chipsalliance/Caliptra/pull/151 to try to clarify this, but not sure if its enough

varuns-nvidia, 2023-10-31:
Thanks Lou, left a few comments. For RTM, I'm of two minds whether the text should be left as-is or should be explicit about RTM of itself vs others.

FerralCoder, 2023-10-31:
Yeah, that's tricky. I hesitated to make changes about RTM as it seems to overly dilute what the promise of Caliptra is

closed 2023-11-01 by andreslagarcavilla