chipsalliance/Caliptra #116: Caliptra.md Error handling & reporting section needs update

FerralCoder, 2023-09-26:
Section is incomplete and no longer well aligned to actual code implementation

https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#error-reporting-and-handling

calebofearth, 2023-09-29:
Recent edits in the error handling section changed the line:
```
When a fatal error occurs, all assets (UDS fuses, DEOBF\_KEK, Key Slots etc.) are cleared. Note that UDS\_FUSE, DEOBF\_KEK may have already been cleared depending on when the fatal error happened.
```
to
```
When a fatal error occurs, all volatile assets that are stored in key vault are cleared.
```

The three fuse registers `internal_obf_key`, `fuse_uds_seed`, and `fuse_field_entropy` are all cleared in hardware when a fatal error is detected, and crypto zeroize functions and Key Vault flush are also triggered.
It sounds like the wording was massaged to refer exclusively to clearing of secrets in VeeR memories (i.e. Firmware), but the original intent was also accurate.

varuns-nvidia, 2023-10-26:
> Recent edits in the error handling section changed the line:
>
> ```
> When a fatal error occurs, all assets (UDS fuses, DEOBF\_KEK, Key Slots etc.) are cleared. Note that UDS\_FUSE, DEOBF\_KEK may have [...]

Key vault is HW right? So the meaning is kept the same.
The reason I made that change was because of inconsistent references to the obfuscation key throughout the spec -- just figured the language could be simplified.

calebofearth, 2023-10-26:
You're right, Key Vault is hardware, so the text is accurate. I think I typed up the bit about FW/VeeR memories in response to [this](https://github.com/chipsalliance/Caliptra/commit/a7bf1f1159ef3b5086cd5e180ad33f9fb23ce884#:~:text=When%20a%20fatal%20error%20occurs%2C%20all%20volatile%20assets%20that%20are%20stored%20in%20memory%20(UDS%20fuses%2C%20DEOBF%5C_KEK%2C%20key%20slots%2C%20etc.)%20are%20cleared.%20Note%20that%20UDS%5C_FUSE%20and%20DEOBF%5C_KEK%20may%20already%20be%20cleared%20depending%20on%20when%20the%20fatal%20error%20happened.) edit and forgot to update it when I realized that wasn't the latest commit.

Thanks for clarifying the motivation here - we still might be able to improve the spec by noting that other assets outside of Key Vault are also cleared on Fatal Errors, but at least this current text is correct.

closed 2023-10-27 by FerralCoder