chipsalliance/Caliptra #159: Boot flow diagram bugs

steven-bellock, 2023-11-15:
In Figure 9 `cptra_uc_rst_b` is deasserted but never asserted. It should be asserted right after `cptra_pwrgood`. 
In Figure 11 `cptra_uc_rst_b` is asserted twice. The second assert should be a deassert.

steven-bellock, 2023-11-15:
@steph-morton how are these diagrams updated? I only see the `.png` files.

steph-morton, 2023-11-15:
There are sources in the shared specification drive. LMK if you need a link.

steven-bellock, 2023-11-15:
> There are sources in the shared specification drive. LMK if you need a link.

Yes please.

FerralCoder, 2023-11-27:
@steven-bellock, are you working to close this issue, or need help from someone else?

steven-bellock, 2023-11-27:
I just got Visio installed so I can work on it this week.

steven-bellock, 2023-11-29:
I talked with @steph-morton and since I am having issues with Visio rendering the diagrams correctly, she will look at it instead.

closed 2023-12-01 by andreslagarcavilla