chipsalliance/Caliptra #173: RISC-V and rust support quality assurance

labba, 2024-01-31:
after reviewing the rust platform supported :
https://doc.rust-lang.org/nightly/rustc/platform-support.html

RISC-V is in the Tier 2 level support and according to the documentation:
1. Tier 2 targets can be thought of as "guaranteed to build"
2. Automated tests are not always run so it's not guaranteed to produce a working build

how was Caliptra decided to be written with a CPU that is not a Tier 1 level CPU  with this level of compiler quality assurance ?

ssarfaty, 2024-03-05:
ping ? anyone can share details ?

labba, 2024-03-05:
any updates ? why not share your decision taking ? this is how "transparency" works ? only one side ?

FerralCoder, 2024-03-22:
Duplicate of https://github.com/chipsalliance/caliptra-sw/issues/1300

closed 2024-03-22 by FerralCoder