chipsalliance/Caliptra #191: Doc inconsistency in UDS vs UDS seed

varuns-nvidia, 2024-05-03:
Should consistently refer to UDS as the output of de-obfuscation and UDS seed as the input.
@FerralCoder , @bluegate010 for thoughts before we jump on this.

FerralCoder, 2024-05-07:
I would agree with that suggestion. @JohnTraverAmd do you agree as well? I do think that we should always refer to the fused value as UDS Seed, and only the value after de-obfuscation as UDS.

That said, the UDS Seed itself should still be generated following rules for generating a NIST compliant UDS.

JohnTraverAmd, 2024-05-13:
I think that the input to the obfuscation should be called obfuscated UDS while the output is called UDS.
UDS Seed may not make sense at this point.  What is the point of creating a term for UDS-Seed?  I think that if anything, we are creating an IDevID Seed.

bharatpillilli, 2024-08-08:
If we call it obfuscated UDS, someone will ask how is Caliptra or manufacturing obfuscating the value (infact some folks were already asking this when we stated UDS deobfuscation). So it is better to keep it as seed.