chipsalliance/Caliptra #192: Potential inconsistencies between fuse map and architectural registers

A-Rovelli, 2024-05-17:
Hello,

We have noticed some differences, or potential inconsistencies, between the fuse map and the corresponding registers, we would need clarification on this. We refer respectively to the tables on the following pages.

Fuse map:
[https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-map](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-map)

Fuse and architectural registers
[https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers)

The first issue is about the size of the _field entropy_:
map: FIELD ENTROPY (obfuscated) **256** bits
reg: FUSE_FIELD_ENTROPY **384** bits

In addition, we would like a confirmation of the fact that for the fields of the fuse whose size is less than 32 (which is the size of the registers implemented in Caliptra) an entire register is still allocated. An example is this:
map: KEY MANIFEST PK HASH MASK 4 bits
reg: FUSE_KEY_MANIFEST_PK_HASH 32 bits

If this is true, however, we believe that in the following case there is an inconsistency:
map: ANTI-ROLLBACK DISABLE 1 bit
reg: FUSE_ANTI_ROLLBACK_DISABLE 1 bit

Finally, the fields IDEVID MANUF HSM IDENTIFIER and SOC STEPPING ID do not have corresponding registers but in fact this does not seem to be necessary, do you confirm that this is the case?

Thanks in advance.

calebofearth, 2024-06-07:
The discrepancies here need to be addressed in the caliptra-sw repo - can you file an analogous issue there to update documentation?

Field Entropy is 256-bits in the hardware
KEY MANIFEST PK HASH MASK is 4-bits
ANTI_ROLLBACK_DISABLE is 1 bit

32-bit registers are the minimum size; a full register is allocated for registers with fewer than 32-bits.
Can you clarify the discrepancy for the ANTI_ROLLBACK_DISABLE field?

Regarding FUSE_IDEVID_MANUF_HSM_ID, please refer to https://github.com/chipsalliance/caliptra-rtl/issues/457

A-Rovelli, 2024-06-11:
Hi,

as per your suggestion:
[https://github.com/chipsalliance/caliptra-sw/issues/1565](https://github.com/chipsalliance/caliptra-sw/issues/1565)

A-Rovelli, 2024-06-11:
> Can you clarify the discrepancy for the ANTI_ROLLBACK_DISABLE field?
I would have expected to see the value 32 written in the "Width (bits)" column at row FUSE_ANTI_ROLLBACK_DISABLE.

varuns-nvidia, 2024-06-13:
1-bit is correct. The HW maps a given logical fuse value to a unique register. But the data is just 1 bit.

bharatpillilli, 2024-08-08:
Seems like all is taken care of - so closing it off

closed 2024-08-08 by bharatpillilli