chipsalliance/Caliptra #194: Doc inconsistency about external vs internal TRNG

A-Rovelli, 2024-05-17:
In Caliptra's documentation, as far as the implementation of TRNG is concerned, it is written that the "internal TRNG" solution is the preferable one, but at the same time it is also written that it is not, at the moment, completely reliable. Since this position seems to be a bit ambiguous, we would like to know what exactly Caliptra WG recommends.

These are the text fragments to which we refer:

[https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#hardware](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#hardware)
_“This mode (HW API) is advised for early development but discouraged for production tape outs due to the lower security assurances of an external TRNG.”_

[https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/CaliptraIntegrationSpecification.md#trng-req-hw-api](https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/CaliptraIntegrationSpecification.md#trng-req-hw-api)
_“While the use of this API is convenient for early enablement, the current Caliptra hardware is unable to provide the same security guarantees with an external TRNG.”_

Moreover, exactly what does the phrase "...is unable to provide the same security guarantees with an external TRNG." mean?

varuns-nvidia, 2024-06-13:
@bharatpillilli @howardtr I think we need to revise this text. iTRNG is the preferred integration solution.

varuns-nvidia, 2024-06-13:
Actually, sorry I read the spec text and it is intentional. The iTRNG is the preferred integration and the spec is consistent. 
> Moreover, exactly what does the phrase "...is unable to provide the same security guarantees with an external TRNG." mean?
It means that because the random number comes from the external APB bus instead of the internal AHB bus, Caliptra cannot provide the same assurances. Hence the recommendation for iTRNG.

bharatpillilli, 2024-08-08:
Closing because there is no more activity on this

closed 2024-08-08 by bharatpillilli