chipsalliance/Caliptra #203: Lifecycle and boot media integrated mode

A-Rovelli, 2024-06-14:
Hi,

in table 19, in the row that describe the field "LIFE CYCLE" the "Caliptra Boot Media Integrated" is mentioned. In my understanding this operating mode is not longer supported in Caliptra 1.0 and I'm wondering if the content of the column description is valid or not. I would need a clarification on that. Thanks.

--
A. Rovelli

varuns-nvidia, 2024-06-14:
Yes it is not relevant for Caliptra 1.0.

A-Rovelli, 2024-06-14:
In other words, just for confirmation, in Caliptra 1.0, that is boot media dependent, I have to take in account of the 2 bit of the LIFE CYCLE field.

varuns-nvidia, 2024-06-14:
Your phrasing is not clear to me. This is what is stated:
> Caliptra Boot Media Integrated mode usage only. SoCs that build with a Boot Media Dependent profile don’t have to account for these fuses.

For Caliptra 1.x, life cycle is encoded in the security state 3-bit input wire, not fuses.

A-Rovelli, 2024-06-14:
> Your phrasing is not clear to me. This is what is stated:
> 
> > Caliptra Boot Media Integrated mode usage only. SoCs that build with a Boot Media Dependent profile don’t have to account for these fuses.
> 
> For Caliptra 1.x, life cycle is encoded in the security state 3-bit input wire, not fuses.

OK, thanks.

closed 2024-06-14 by varuns-nvidia