chipsalliance/Caliptra #204: Clarification on "redundant fuses" expectation ?

myviewfinder, 2024-07-05:
Is the text in Caliptra section below describing Caliptra's internal "redundant fuses" beyond the fuse map registers?  Are there any expectation on the SoC about implementing and/or programming "redundant fuses" into Caliptra fuse map registers?

Caliptra specification section [Device Resilience](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#device-resilience) Table 12: NIST SP 800-193 requirements, row number 4.3.2 contains description below:

> Caliptra relies on redundant fuses to store its configuration data, which is owned and passed to Caliptra through the Mailbox.

varuns-nvidia, 2024-07-05:
No, the expectation is entirely on the SoC. Caliptra IP does not implement fuses. See the following text from https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-requirements

> SoC should ensure that the integrity of each fuse is maintained through the life of the part. The integrity of the fuses can be maintained by fuse redundancy, ECC, or other means determined sufficient by the SoC.

myviewfinder, 2024-07-09:
Kindly consider Caliptra specification clarification:

1. Please consider using consistent language in the Caliptra specification section [Device Resilience](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#device-resilience) Table 12: NIST SP 800-193 requirements, row number 4.3.2,   with the section  [Fuse Requirements](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-requirements), and

2. Replace the "through the Mailbox" text to "through fuse registers".   SoC writes Caliptra fuses via Caliptra fuse registers, not Caliptra Mailbox.

The current sentence in specification section Device Resilience:
> "Caliptra relies on redundant fuses to store its configuration data, which is owned and passed to Caliptra through the Mailbox."

An example after combining the bullet list above:
> "Caliptra relies on _SoC fuse integrity_ to store its configuration data, which is owned and passed to Caliptra through the _fuse registers_."

varuns-nvidia, 2024-07-09:
Thanks for the suggestion, PR posted.

commit f7adc06f (2024-07-09): Update use of term "redundant fuses"

Fixes #204
PR #207 (2024-07-18): Update use of term "redundant fuses"
Fixes #204
commit 1317e003 (2024-07-18): Update use of term "redundant fuses" (#207)

Fixes #204
closed 2024-07-18 by andreslagarcavilla