chipsalliance/Caliptra #205: Fuse "IDEVID CERT IDEVID ATTR" and IDevID Certificate incomplete correlation, discrepancy and typo.

myviewfinder, 2024-07-05:
Summary:  Caliptra main specification and Caliptra Integration spec. do not fully define the fuse "Certificate Attribute" fields with corresponding fields in the IDevID Certificate.    There is also typo in the Caliptra ROM spec.  Details below.

1. What does fuse Certificate Attribute "Flags (byte 0, bits [1:0]): Key ID algorithm for IDevID Subject Key Identifier" correspond to in the [IDevID Cert](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#idevid-certificate) ?   I.e. is Table 7: IDevID certificate missing the "Subject Key Identifier" field?

2. What should fuse Certificate Attribute "Subject Key ID (bytes 4 to 23) " be when fuse Certificate Attribute's Flags != 3?

3. What is fuse Certificate Attribute "Manufacturer Serial Number (bytes 28 to 43):" used for?  Table 7 does not have such a IDevID certificate field. 

4. There are discrepancies in the Certificate Attributes (i) fuse size and (ii) location of fields among various specification.  Main spec [encoding of these attribute fuses below table 7](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#idevid-certificate) defines attributes **= 44 bytes**, versus 

> - the followings with **96 bytes**:  [Fuse Map table 19](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-map) with 768 bits **= 96bytes**  (https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#fuse-map), 
> - ereg fuse register [fuse_idevid_cert_attr  fuse array size = 0x60](https://chipsalliance.github.io/caliptra-rtl/main/external-regs/?p=caliptra_top_reg.generic_and_fuse_reg.fuse_idevid_cert_attr%5B0%5D) ,  and 
> - [ROM spec. 768 bits](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers) 

5. Typo in [ROM spec FUSE_IDEVID_CERT_ATTR](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers) with same encoding value of "2" for both SHA256 and SHA384 selections.

6. [ROM spec FUSE_IDEVID_CERT_ATTR](https://github.com/chipsalliance/caliptra-sw/blob/main/rom/dev/README.md#fuse-and-architectural-registers) fields & locations discrepancies with that in the [Caliptra main spec.'s encoding of these attribute fuses below table 7](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#idevid-certificate)

varuns-nvidia, 2024-07-05:
1. The fuse IDEVID_CERT_ATTR bits 0 and 1 tell Caliptra ROM what is the format and the input data for the IDevID subject key identifier. Yes you're right that Table 7 should include the Subject Key Identifier.
2. The value is don't care because the Flags tell Caliptra ROM not to consume that value.
3. It is used to generate the tcg-dice-Ueid extension value.
4. The fuse bits allocated in the Caliptra fuse map is 96 bytes = 768 bits. Of those, only 44 bytes have defined usage.
5. @mhatrevi please review
6. @mhatrevi please review

varuns-nvidia, 2024-07-05:
I've pushed https://github.com/chipsalliance/Caliptra/pull/206 with the main spec updates adding Subject Key Identifier.

myviewfinder, 2024-07-09:
> 1. The fuse IDEVID_CERT_ATTR bits 0 and 1 tell Caliptra ROM what is the format and the input data for the IDevID subject key identifier. Yes you're right that Table 7 should include the Subject Key [...]

@varuns-nvidia , 
About your response to bullet no.2, please kindly clarify in the Caliptra Main specification that SoC shall treat it as Don't Care in byte(4 to 23) when Flags != 3 .  

Additional issue on bullet no.3, does the text below miss out listing the **IDevID**, since IDevID cert. field also has tcg-dice-Ueid field?
>"Manufacturer Serial Number (bytes 28 to 43): the 128-bit unique serial number of the device to be used for the TCG UEID extension in the Caliptra-generated LDevID, AliasFMC, and AliasRT certificates." 

About your response to bullet no.4, please consider the following clarification in the Caliptra Main spec. [encoding of certificate attribute fuses below table 7](https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#idevid-certificate):  
> Reserved (byte 44 to byte 95)

7. In general does Caliptra have an expectation on any _reserved_ field value, such as 0 ?