chipsalliance/Caliptra #216: What is RWL-S at register access comment

byunggu-jo, 2024-08-01:
Hello,

I want to know meaning of RWL-S at register access comment
(https://chipsalliance.github.io/caliptra-rtl/main/internal-regs/?p=clp.soc_ifc_reg.fuse_soc_stepping_id)

I understand abouot RW, RO, WO but can't understand "L-S".
Please help me.

calebofearth, 2024-08-01:
The interpretation for this access is:
"Read/Write Lock - Sticky" => RWL-S
This means that this register is writable by the SoC until [CPTRA_FUSE_WR_DONE](https://chipsalliance.github.io/caliptra-rtl/main/internal-regs/?p=clp.soc_ifc_reg.CPTRA_FUSE_WR_DONE) has been set. After that, the value written persists until a cold reset (cptra_pwrgood is deasserted).

closed 2024-08-01 by calebofearth

byunggu-jo, 2024-08-02:
Thank you for response.

I have one more quesetion.
SoC means HW or include SW?
At your response and document of Caliptra, I can find "SoC should write ..." comment.
So, I want to know the SoC means other HW logic (not include FW) or SoC with out caliptra any thing (include other security CPU)
(https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/CaliptraHardwareSpecification.md)

calebofearth, 2024-08-02:
SoC is any logic outside of Caliptra. The implementation may include HW or ROM logic, but the typical design would be to implement this as FW (ROM) in a security processor that is outside of the Caliptra boundary.
See [SoC Integration Requirements](https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/CaliptraIntegrationSpecification.md#soc-integration-requirements):
> SoC shall write fuse registers and fuse done via immutable logic or ROM code.