chipsalliance/Caliptra #264: 64-bit core option?

solomatnikov, 2025-02-05:
Any plans for a 64-bit processor core option?

jhand2, 2025-02-08:
What would be the use-case? Caliptra doesn't have that much addressable memory.

solomatnikov, 2025-02-09:
Caliptra (and MCU in particular) is connected to the "Rest of SOC":

[image]

SoC and multi-chiplet design physical address space can easily exceed 4 GB.

One of advantages of RV was and still is availability of 64-bit microcontroller cores, which can address > 4 GB directly, plus higher performance of 64-bit cores. Area/gate count difference between 64-bit and 32-bit microcontroller cores is small/negligible.

This is why RV 64-bit microcontroller cores became popular in SoC designs.

jhand2, 2025-02-10:
Ah, so this question is specifically for subsystem. + @bharatpillilli

solomatnikov, 2025-02-13:
I can take a look at that next month or in April.

I think it would be cleaner and easier if both cores are switched to 64-bit.

How do you envision implementation of 64-bit core option?

Is it important to have hypervisor support in these cores?