chipsalliance/Caliptra #267: No standard protocol/api definition provided by Caliptra for in-field eFuse programming.

rstrongMCT, 2025-02-07:
Caliptra 1.x and 2.0 define eFuse that are expected to be in-field programmable. The specification further states that the SoC shall implement an authorization scheme to mitigate DoS attacks and other threats.

However, the specifications states: "Authorization design is outside the scope of this specification" 

It seems that by not defining or establishing standardized requirements to how the authorization mechanism shall be implemented, including the definition of any parameters/payloads, will lead to a plethora of customized solutions (e.g., vendor specific commands) which may result is confusion in the ecosystem causing a device vendor to implement solutions on a per customer basis.

It seems that the mutual authentication features of SPDM 1.4 would provide that standard. In which case, the Caliptra WG would work to establish standard parameter and payload definitions necessary to securely program eFuses. 

It is highly recommended that the WG consider creating a project to address this.

raghuncstate, 2025-02-08:
FYI, the SPDM Workgroup is working on an authorization specification, also been introduced at OCP:

https://www.dmtf.org/content/now_available_authorization_wip_presentation_released_spdm_working_group
https://www.dmtf.org/sites/default/files/standards/documents/DSP0289_1.0.0WIP80.pdf

Should be able to use the authz mechanisms being defined above (note WIP status of spec) and define your parameters/payloads here, or at OCP, or perhaps even take it to SPDM workgroup to standardize as a DMTF authorization profile.

rstrongMCT, 2025-02-10:
I believe that Caliptra is the owner of this since the eFuse map is defined per Caliptra Core specification. Maybe OCP is preferred if this offers the quickest path to definition. Good WG discussion.

rstrongMCT, 2025-04-15:
per discussion, implementation of this needs to be per customer requirements.

rstrongMCT, 2025-04-15:
Reopening this Issue as this question is related. To recap:

In the 10 April meeting, it was discussed that the authorization scheme used to support in-field programmable eFuse should be based on customer requirements.  No issues.

Use of an authorization scheme makes sense for the Entropy values as these are to be provisioned by the customer. However, is this also true for the other in-field programmable values? 

The Caliptra Subsystem Hardware Specification Version 0.5 (yes, understood that https://github.com/chipsalliance/caliptra-ss/blob/main/docs/Caliptra%202.0%20Subsystem%20Specification%201.pdf is superseded by  https://github.com/chipsalliance/caliptra-ss/blob/main/docs/CaliptraSSHardwareSpecification.md) has some details regarding IFP:

1. Any IFP fuse(s) that needs to be provisioned can come over in-band interface or out-of-band interface.
  a. In-band interface is SOC specific
  b. Out-of-band interface will use MCTP commands.
2. IFP payload is issued to MCU;
3. IFP payload must be signed and must be authenticated by Caliptra.
4. MCU will execute SOC specific fuse “burn” flow (which typically involves
regulating an external/internal VRs or LDOs etc. and executing some checks)

Was this abandoned under the "Authorization design is outside the scope of this specification." statement? In which case MCU SDK will not provide any base/reference code and it's up to (based on customer requirements) for the device vendor to develop the appropriate solution based on customer input.