chipsalliance/Caliptra #270: Recovery timeouts?

rstrongMCT, 2025-02-13:
Does Caliptra implement timers used to monitor streaming boot? E.g., on power on, does Caliptra wait forever for its FMC/RT FW? Same question regarding waiting for MCU code and SoC code.

jhand2, 2025-02-26:
Caliptra core has a watchdog timer configured by the SoC via fuse registers. If Caliptra ROM does not boot to RT firmware before the timeout, the watchdog will fire and Caliptra FW will NMI and emit an error.

cc @swenson @mhatrevi to answer for MCU behavior.

rstrongMCT, 2025-04-03:
In addition to MCU RT, what about SoC FW?  Trying to understand what cases could result device due to failure by the BMC to deliverr all FW (Caliptra, MCU and SoC) components

jhand2, 2025-04-07:
SoC FW (or SoC fusing) should configure the timeout to account for potential timeouts/delays related to streaming boot.

If BMC takes too long to stream firmware and caliptra FW is not loaded before WDT fires, then yes, Caliptra will NMI and be unresponsive until next cold reset.

rstrongMCT, 2025-05-28:
I do not see information related to eFuse registers. https://github.com/chipsalliance/caliptra-ss/blob/main/docs/CaliptraSSHardwareSpecification.md#watchdog-timer is silent regarding this as is https://github.com/chipsalliance/caliptra-mcu-sw/blob/main/docs/src/rom.md

zhalvorsen, 2025-07-11:
The units are in Caliptra Core clock cycles

rstrongMCT, 2025-07-14:
and, is it anticipated that MCU FW will set the WDT to a value that is different than what is required for boot? It seems that the boot time timeout will be much longer that non-boot timeouts.