chipsalliance/Caliptra #276: Dcumentation error

rstrongMCT, 2025-02-25:
In https://github.com/chipsalliance/caliptra-sw/blob/main-2.x/rom/dev/README.md, Manufacturing Flows->Debug Unlock, there appears to be documentation error as it is stated:

3. Upon receiving the TOKEN command, ROM sets the CPTRA_DBG_MANUF_SERVICE_RSP_REG register MANUF_DEBUG_UNLOCK_IN_PROGRESS bit to 1.

4. The ROM then constructs a token by prepending and appending the 128-bit value with two 64-bit zeroed values:
64-bit 0s || 128-bit value || 64-bit 0s

5. The ROM then appends a 256-bit random nonce to the token and performs a SHA-512 operation to generate the expected token.

6. The ROM reads the value from the MANUF_DEBUG_UNLOCK_TOKEN fuse register and applies the same transformation as **steps 3 and 4** to obtain the stored token.

Believe that step 6 needs to state: "The ROM reads the value from the MANUF_DEBUG_UNLOCK_TOKEN fuse register and applies the same transformation as **steps _4_ and 5 (using the same 256-bit random nonce)** to obtain the stored token.

jhand2, 2025-02-26:
> Believe that step 6 needs to state: "The ROM reads the value from the MANUF_DEBUG_UNLOCK_TOKEN fuse register and applies the same transformation as steps 4 and 5 (using the same 256-bit random nonce) to obtain the stored token.

Agreed. Would you be willing to send a PR for this?

> Based on the current documentation, it seems that the mailbox scheme for FW downloading, is only applicable to Passive mode? Subsystem Boot Flow has no mention of a mailbox. In which case, https://github.com/chipsalliance/caliptra-sw/blob/main-2.x/rom/dev/README.md (Downloading firmware image from mailbox) needs to be reconciled as it states: "Following is the sequence of steps that are performed to download the firmware image into the mailbox in ACTIVE mode."

In both active and passive mode, Caliptra firmware is loaded via the mailbox. The main difference is how that loading happens. In passive move, it is just via a normal mailbox command that sends the whole image.

In active mode, fw load is invoked with a mailbox command and then Caliptra will DMA the image from the recovery interface into the mailbox.

So I think the existing documentation is correct. Is there still something you think could be clarified more?

rstrongMCT, 2025-02-27:
Thanks for the response. regarding the mailbox confusion - the explanation helps.  I will be useful to be consistent in use of terminology. 'Active' mode seems to be an artifact from Caliptra 1.x whereas 'streaming boot' is the current terminology described by Caliptra 2.x. It is a nit, but the reader (like me) may be unsure as to whether the documentation is the latest and applicable to 2.x.

thanks

closed 2025-04-15 by rstrongMCT