chipsalliance/Caliptra #277: Documentation omission

rstrongMCT, 2025-02-28:
In https://github.com/chipsalliance/caliptra-ss/blob/main/docs/CaliptraSSHardwareSpecification.md#caliptra-subsystem-lcc-state-definitions, there is mention of the RAW_UNLOCK token. 

This token is not described in the eFuse but it is mentioned as a note in https://github.com/chipsalliance/caliptra-ss/blob/main/docs/Caliptra%202.0%20Subsystem%20Specification%201.pdf:

Notes:
- Some tokens are hardcoded design constants (specifically for RAW to
TEST_UNLOCK), while others are stored in FUSE.

This note should be added to the page.

Is it correct to assume that that https://github.com/chipsalliance/caliptra-ss/blob/main/docs/CaliptraSSHardwareSpecification.md#caliptra-subsystem-lcc-state-definitions is the source of truth and the pdf version is to be ignored?

JohnTraverAmd, 2025-04-02:
@calebofearth , Could you look at this documentation on respond?

calebofearth, 2025-04-02:
I've asked @ekarabu to take a look

calebofearth, 2025-04-03:
@rstrongMCT A general response to your question: Yes, the Markdown document that you linked is more up-to-date (0p8 version) and supersedes the PDF version (0p5 version). Future spec updates will be made to the Markdown doc. We will be removing the PDF version soon.
Emre can address questions you have related to constant tokens VS fuse tokens. We'll close this issue out once the central spec is updated with such info.