chipsalliance/Caliptra #281: Clarification on "External-TRNG REQ HW API" steps

jchanga, 2025-03-05:
Hello,

Can you clarify the actual operation of the step 1? Does it mean Caliptra FW writing 1 to register CPTRA_TRNG_STATUS.DATA_REQ? 

1. Caliptra asserts TRNG_REQ wire (FW made the request for a TRNG). 
2. SoC writes the TRNG architectural registers.
3. SoC writes a done bit in the TRNG architectural registers.
4. Caliptra desserts TRNG_REQ.

https://github.com/chipsalliance/caliptra-rtl/blob/main/docs/CaliptraHardwareSpecification.md#external-trng-req-hw-api

Thanks,
Jonathan

jchanga, 2025-05-12:
In addition, can you elaborate exactly how is the data written into TRNG architecture registers used internally? Just to eliminate speculations.

bharatpillilli, 2025-05-21:
eTRNG is unused for Subsystem mode.

For #1 step, this is internal implementation of Caliptra and you will see a wire getting asserted.

For ITRNG, it has a noise source input that SoC provides and the block is internal to Caliptra. Since the logic is open, you can find the iTRNG registers and implementation open for you to audit to eliminate any possible confusions. It's a block that has been reused from OT which has been proven on silicons

closed 2025-05-22 by jchanga