chipsalliance/Caliptra #299: FW Image Verification

rstrongMCT, 2025-03-19:
Would like to get clarification of how Caliptra perform verification of FW images.

In https://github.com/chipsalliance/caliptra-mcu-sw/blob/main/docs/src/image_loading.md, it is stated:

21. MCU RT reads a chunk of the image into a local buffer and writes it directly the image to the target load address. (In the example custom SOC design, this will be the Vendor RAM or Vendor Cfg Storage). This is done until all chunks are copied to the destination.
22. MCU RT sends a Caliptra mailbox command to authorize the image in the SHA Acc identified by the image_id in the image metadata.
23. Caliptra RT sends the image to the SHA Acc.
24. Caliptra RT verifies the computed hash in SHA acc versus the one in the SOC manifest corresponding to the image_id given.
25. Once verified, Caliptra RT returns Success response to MCU via the mailbox. AUTHORIZE_AND_STASH
The SoC uses this command to request authorization of its various SoC images. This command has the option to receive the image hash directly from SoC or from an external source (e.g., SHA Acc).

Step 23 implies that it is Caliptra RT (and not SoC/MCU) that requests that a SHA digest be created? This means that SoC/MCU does NOT use the CM_SHA_INIT, SM_SHA_UPDATE and CM_SHA_FINAL MB commands. That seems to make sense as the latency associated with computed SHA across a large FW image would not be tolerable. 

But, it is stated in :: https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md

Common Run-time Authentication Flows

1. MCU RT FW will do PLDM T5 flow to obtain the FW images for downstream uControllers (or other SoC configuration)
2. MCU RT FW will send/stream the (FW or config) payload to Caliptra SHA Acc to perform hash measurements as the payload comes through the MCTP transport.

This implies that MCU/SoC is required to explicitly use some Caliptra mechanism (CM_SHA_XXX MB commands?) to compute the SHA digest. 

Which of these is correct?

JohnTraverAmd, 2025-04-02:
I think this diverged based on the changes coming in with Caliptra 2.0.  @mhatrevi, can you look at this documentation to align it?

swenson, 2025-04-03:
The SHA accelerator peripheral is no longer directly accessible outside of the Caliptra core, so the SHA mailbox commands will be used to compute the SHA.

rstrongMCT, 2025-04-03:
The SHA MB commands are the CM_SHA_XXX commands, correct? Any performance data available?

swenson, 2025-04-09:
@rstrongMCT Correct. We don't have specific performance numbers. The CM_SHA_XXX commands can process 4096 bytes at a time, and the bytes are copied (4 at a time) into the SHA accelerator peripheral, which is then pretty fast.

rkg137, 2025-04-10:
[image]

CB commands uses mailbox to transfer the data, Performance expected to be low.
If image size required to be authorized is in MBs, it will introduce alot of latency to compute the digest of the Image before AUTHORIZE_AND_STASH command can be issued.

I understand we don't have performance data for SHA Acc, 
Do we have performance data for mailbox data transfer(128KB,256KB,1MB etc).?
Do we have any other interface to compute the digest where SoC can just provide the Its own SRAM address and SHA Acc(DMA widget) can stream through data from SoC SRAM. 

Can you help to clarify on the below, if this can be utilized for digest calculation.?
"Hash calculation HW API (Subsystem mode only)"

[image]

closed 2025-04-15 by rstrongMCT