chipsalliance/Caliptra #315: Passive Mode v. Subsystem Mode

rstrongMCT, 2025-04-02:
In https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md, it is stated "From Caliptra 2.x onwards, Caliptra introduces two modes of operation. Passive mode which was supported in 1.x architecture and Subsystem mode.'

"Caliptra is among the first microcontrollers taken out of reset by the power-on reset logic. Caliptra coordinates the start of the firmware chain-of-trust with the immutable component of the SoC ROM. After the Caliptra ROM completes initialization, it provides a "stash measurement" API and callback signals for the SoC ROM (passive mode) to proceed with the boot process"

The spec also states:
Subsystem Boot Flow
If (Caliptra-Passive-Mode)
                   SoC Manager goes through Caliptra 1.x flows => loads Caliptra FW using Caliptra 1.x flows, Caliptra sets RT 
                    ready and SOC <-> Caliptra boot flow is done.
(Caliptra-Subsystem-Mode)
...

This seems to indicate that passive mode is not possible when subsystem is present?

JohnTraverAmd, 2025-04-09:
Correct. Passive mode and subsystem mode are two mutually exclusive configurations.

JohnTraverAmd, 2025-04-09:
I'm closing this. Please re-open if there is something actionable you need from us.

closed 2025-04-09 by JohnTraverAmd