chipsalliance/Caliptra #320: Caliptra fatal error handling

rstrongMCT, 2025-04-08:
In https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md, Error reporting and Handling, it is stated:

"Caliptra signals fatal errors using a cptra_error_fatal wire.

- SoCs should connect this into their SoC error handling logic. Upon detection of a FATAL ERROR in Caliptra, SoC shall treat any outstanding commands with Caliptra as failed, and SoC may recover by performing a Caliptra reset using the signal cptra_rst_b.
 
- This signal prevents forward progress of the boot process if measurement submission to Caliptra fails. 

What does forward progress mean in this context? After signaling a fatal error via cptra_error_fatal wire, what does Caliptra do? Does Caliptra enter a halt state until the SoC resets it via cptra_rst_b? 

A fatal error condition can be "Firmware image verification or authentication failures if encountered by ROM during Cold Reset flow."  Which FW images? Does this include any authentication failures detected for Caliptra FMC, Caliptra RT, SoC and MCU manifests, MCU RT and SoC FW?  In which case, what does MCU do if SoC FW has an authentication failure?

bharatpillilli, 2025-05-21:
This is SoC specific architecture on how you will handle a fatal error for your SOC and your existing RoTs.

Please check with your SoC arch.

As documented, Caliptra will hang on a fatal error and only a reset of it will bring it out and if those errors are due to double bit ECC only a cold reset will as documented in the spec.

rstrongMCT, 2025-05-27:
[caliptra-ss/docs/CaliptraSSIntegrationSpecification.md at main · chipsalliance/caliptra-ss](https://github.com/chipsalliance/caliptra-ss/blob/main/docs/CaliptraSSIntegrationSpecification.md#caliptra-subsystem-error-and-recovery-flow) states that "In the event of a non-fatal error during boot, SoC should enter recovery flow and attempt to boot again using alternate boot part/partition."
- Q: In a streaming boot environment, what is an "alternative boot partition". Is this managed by BMC or equivalent?

It is also stated "if SoC sees that a non-fatal error has occurred AFTER initial bringup, SoC may attempt to recover Caliptra by executing the “Run Self-Test” mailbox command (not yet defined).

Q: "AFTER initial bring up" means that Caliptra and SoC (MCU + other SoC IPs) have successfully booted? But perhaps not based on this statement: "If this command is unsuccessful, Caliptra is in an error state for the remainder of the current boot. "?
Q: If Caliptra is in an error state, what Caliptra functionality is lost? Is the SoC considered to be running in an inscure state?
Q: when will the Run Self-Test mailbox command details be available?

bharatpillilli, 2025-05-27:
!!!!Great Questions!!!!

"It is also stated "if SoC sees that a non-fatal error has occurred AFTER initial bringup, SoC may attempt to recover Caliptra by executing the “Run Self-Test” mailbox command (not yet defined)." -> this wording came from 1.x and chasing down the relevancy of this as there is no such mailbox command implemented AFAIK. And subsystem copy/pasted the entire section from Core spec. 

"[caliptra-ss/docs/CaliptraSSIntegrationSpecification.md at main · chipsalliance/caliptra-ss](https://github.com/chipsalliance/caliptra-ss/blob/main/docs/CaliptraSSIntegrationSpecification.md#caliptra-subsystem-error-and-recovery-flow) states that "In the event of a non-fatal error during boot, SoC should enter recovery flow and attempt to boot again using alternate boot part/partition."" -> The whole point of streaming boot is do away with recovery flow. 

This section certainly needs some clean up in both Caliptra core and subsystem. Caleb will look at it once he is back from vacation.

calebofearth, 2025-06-26:
PR's addressing the referenced integration spec sections have been merged.
As the error handling flow was an example/reference flow only, it has been replaced with pointers to the main spec (https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#error-reporting-and-handling) for clarity.

rstrongMCT, 2025-07-14:
@bharatpillilli @calebofearth:

We believe that a failed FW verification (resulting in a fatal error condition) will only be applicable to Caliptra FMC and RT. Since a fatal error will be thrown by Caliptra, SoC can detect this and perform a reset of Caliptra. Expect that Caliptra ROM will report the failure to BMC.

If Caliptra RT fails to verify MCU FW, then we will expect Caliptra RT to handle this and report an error to BMC. This is not fatal. MCU FW and SoC FW are not running and therefore are not involved in FW download retries or other activities.

If MCU FW fails to verify SoC FW (via Caliptra) components, then this is not considered fatal. MCU FW should fail the PLDM authorize command. SoC FW is not running, and therefore are not involved in FW download retries or other activities.

Is this our understanding accurate?

What is not well described, is what functionality will Caliptra lose if a non-fatal error occurs. Are there measures that MCU/SoC must take in order to get the non-fatal error conditions to clear? For example, Cryptography processing errors are considered non-fatal as well as some mailbox errors; if detected does the device enter a cripple state?