chipsalliance/Caliptra #324: Stream Data from SoC/MCU SRAM to SHA Acc - through AXI

RaunakGu, 2025-04-10:
Is there a way where the SoC/MCU can act as an initiator, providing its own SRAM address, and the SHA Accelerator (DMA widget) can stream data from SRAM through AXI DMA transfer to compute the digest?

Seems from code that Caliptra FW can do this, but no interface exposed for MCU/SoC to trigger this ?

[image]

swenson, 2025-04-10:
The MCU does not have access directly to the SHA accelerator in 2.0. I'm not sure if it can program the DMA widget to access it or not.

The more straightforward way would be to use the cryptographic mailbox SHA commands: https://github.com/chipsalliance/caliptra-sw/tree/main-2.x/runtime#mailbox-commands-cryptographic-mailbox-20

RaunakGu, 2025-04-10:
Appreciate the reply,

Yes, I understand MCU doesn't have access to the SHA Acc, SHA Acc is part of Caliptra core.

SHA MB commands required data to be passed through mailbox, this will have performance impact,

[image]

I was expecting "Can MB commands be used to optionally pass the SRAM address for complete data through the mailbox, allowing the Caliptra Core to stream the data from SRAM to the SHA Accelerator via AXI DMA transfer?"

bharatpillilli, 2025-05-21:
> Appreciate the reply,
> 
> Yes, I understand MCU doesn't have access to the SHA Acc, SHA Acc is part of Caliptra core.
> 
> SHA MB commands required data to be passed through mailbox, this will have performance impact,
> 
> <img alt="Image" width="242" src="https://private-user-images.githubusercontent.com/53434307/432467677-9e899d01-9fb6-4548-b622-7f4a3b9812df.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDc4NDI4NTYsIm5iZiI6MTc0Nzg0MjU1NiwicGF0aCI6Ii81MzQzNDMwNy80MzI0Njc2NzctOWU4OTlkMDEtOWZiNi00NTQ4LWI2MjItN2Y0YTNiOTgxMmRmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA1MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNTIxVDE1NDkxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZjY2M0NTIyN2IwZDRhNjc4YTQ4Mzg5ZjE1MmQ1OTY1YmY1ODYzYzdmMjg4NGZkZjFlOTk1ZTQ5NzhiYmZmZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FsNd_2yP5E4pLuhvGf1lZZFGbLSjp9js1YF9k1BB5eE">
> I was expecting "Can MB commands be used to optionally pass the SRAM address instead of complete data through the mailbox, allowing the Caliptra Core to stream the data from SRAM to the SHA Accelerator via AXI DMA transfer?"

@swenson can answer but HW capability exists

RaunakGu, 2025-09-17:
This has been resolved, closing it