chipsalliance/Caliptra #352: Caliptra Main spec missing updated eFuse requirements

rstrongMCT, 2025-05-14:
eFuse map mismatch inCaliptra/doc/caliptra_20/Caliptra.md at main · chipsalliance/Caliptra and caliptra-ss/src/fuse_ctrl/doc/otp_ctrl_mmap.md at main · chipsalliance/caliptra-ss

Caliptra main states:
Fuse map
FIXME: Needs updates for Caliptra 2p0 & Subsystem The following table describes Caliptra's fuse map:

Unclear if the flow described in https://github.com/chipsalliance/Caliptra/blob/main/doc/caliptra_20/Caliptra.md#firmware-image-format-and-verification is accurate based on the discrepancies (e.g., references to key_manifest_pk_hash??)

bharatpillilli, 2025-05-22:
Main spec carries only Caliptra core. Please refer to the subsystem spec for fuses outside of Caliptra core. Reason for this division is what Subsystem requires is SOC centric architectural decisions (ex: debug unlock hash keys, caliptra FW auth keys, certs etc.)

Caliptra main spec ONLY carries caliptra core fuses that are mandatory.

rstrongMCT, 2025-05-22:
from a readability perspective, it would be helpful to a note/link to the other pertinent sections because as written (The following table describes Caliptra's fuse map:), it would seem that the specified fuse table is the source of truth. The FIXME note in main needs to be updated/removed as the reader may infer that the updates, when they occur, will impact the described flows.

rstrongMCT, 2025-05-27:
Please make necessary updates or indicate if no FIXME is required. Caliptra main states:

Fuse map
FIXME: Needs updates for Caliptra 2p0 & Subsystem The following table describes Caliptra's fuse map:

rstrongMCT, 2025-07-14:
Are there any additional modifications planned to the fuse map described in Caliptra Main? If not, can the spec be updated to remove the FIXME? This is causing some uncertainly .

closed 2025-10-24 by rstrongMCT