chipsalliance/Caliptra #374: Open on CSR HMAC signing on IDevID CSR TBS

Alexandiz, 2025-05-29:
Assumption Caliptra ROM is responsible to generate IDevID CSR TBS and to perform or not the CSR HMAC signing based on MAC512_CTRL.CSR_MODE register setting.

Regarding the usecase of manufacturing with CSR HMAC flow - the open is how Caliptra ROM or MCU ROM can identify if CSR HMAC signing is needed and set MAC512_CTRL.CSR_MODE register accordingly while it is per-project decision.

The CSR HMAC key in Netlist and CSR HMAC signing can be in the following states: 
1. The CSR HMAC key is not embedded in the Netlist, the CSR HMAC signing is disabled. 
2. The CSR HMAC key is embedded in the Netlist, but CSR HMAC signing is disabled in manufacturing facility. 
3. The CSR HMAC key is embedded in the Netlist and CSR HMAC signing is enabled in manufacturing facility. 

Based on this, I assume that Caliptra ROM or MCU ROM can't know the status of embedding CSR HMAC key in Netlist and can't know w/o external setting if CSR HMAC signing is enabled in manufacturing facility. I assume that such external setting can be configured in Netlist (less flexible, thus not sufficient option) or Fuse or HVM command over JTAG. 

I am referring to the IDevID CSR TBS flow 
https://github.com/chipsalliance/Caliptra/blob/main/doc/Caliptra.md#provisioning-idevid-during-manufacturing
Thank you

+ @bharatpillilli

Alexandiz, 2025-06-25:
@JohnTraverAmd  and @varuns-nvidia ,
can you comment on this issue? 
Thanks,
Alex

kgotze, 2025-06-25:
hi @Alexandiz and @bharatpillilli  ... commenting at the behest of Varun:

is there a reason why the HMAC computation needs to be conditional at all?

going through the enumerated cases:

_1) The CSR HMAC key is not embedded in the Netlist,_

         clearly there is no intent to validate HMAC, the garbage HMAC (generated with 0 / X) can be ignored. 

_2) The CSR HMAC key is embedded in the Netlist, but CSR HMAC signing is disabled in manufacturing facility._

          the generated MAC that's returned can be easily ignored in the tester/ MFG infrastructure.

_3) The CSR HMAC key is embedded in the Netlist and CSR HMAC signing is enabled in manufacturing facility._

         the generated MAC can be extracted through the mailbox and used for CSR authentication and subsequent issuance.

if there's a need, I don't think we'd object to a tie-up/tie-down to enable CSR generation on an embedding basis. (or some kind of tester driven JTAG write if absolutely required) but in the spirit of "keep it simple" I'd ask why do you need this to be conditional at all?

Alexandiz, 2025-06-26:
Hi @kgotze,
In this case the MAC512_CTRL.CSR_MODE register shall be always set to generate CSR HMAC signing (even if key is not valid or CSR HMAC is not requested). So it will become per-project decision in manufacturing facility if to use it or not.
I assume this direction will impact MCU.ROM to always set MAC512_CTRL.CSR_MODE register.
I suggest to document these requirements. 
Thank you,
Alex

varuns-nvidia, 2025-06-26:
@mhatrevi @swenson could you document this need for MCU ROM?