chipsalliance/Caliptra #489: Caliptra FW authentication related questions

sshar14, 2025-07-28:
Hi,
 
We have a few questions related to caliptra firmware authentication, signing and security version.
1.	When caliptra firmware image contains a vendor signature but lacks an owner signature, this could be the case in an owner non-production / development environment. what is the expected behavior for caliptra firmware image authentication?
2.	Given that caliptra does not support selective enablement of PQC and classical signing for firmware images, how should scenarios be handled where the owner lacks PQC infrastructure or intends to exclusively use PQC signing?
3.	What is the recommended method to enforce that older or specific firmware version cannot run on newer or different caliptra hardware revisions? Who is responsible for managing the SVN values and is there a flow describing this?
 
Thanks
Shalini

jhand2, 2025-07-28:
1. Caliptra FW must always have an owner signature. In non-prod/development environment, the expectation is that the owner pubkey hash register is set to all 0s. This will allow FW signed with any key. In this case, it is up to you how to handle this (e.g. sign with the same key as vendor key, sign with a test key)
2. @mhatrevi can you comment on this?
3. Recommendation is to use the `fuse_fmc_key_manifest_svn` and `fuse_runtime_svn` fuses. Caliptra ROM will use these to enforce anti-rollback. The flow for blowing these fuses is handled by SoC Manager (or MCU in Subsystem mode) to ultimately it is up to the integrator. We do not currently have a recommended design, but I think it would be reasonable for the Caliptra Workgroup to provide a recommendation. We can discuss this and get back to you with more information.

ismaws, 2025-08-13:
Follow up questions:

1: is the behavior also applicable to the manufacturer signature? i.e. ROM will load FMC or RT signed with any key if manufacturer keys is set to al 0s?

2: any open source tool recommendation to sign firmware with PQC algs? I'm still not clear from your response if PQC signatures are mandatory or optional i.e. firmware will not be loaded if no PQC signature is present?

3: "The flow for blowing these fuses is handled by MCU in Subsystem mode" ... what MCU code? ROM? FMC? RT? Any? i.e. is MCU completely trusted / in the TCB from Caliptra's core pov?